py32duino/Arduino-PY32 #1: Example request

garudaonekh, 2023-11-19:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

No

### 描述您想要的解决方案 / Describe the solution you'd like

Please provide Arduino example so that it is easy to get started.

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

Example and configuration

### 确认信息

- [X] 检查过该需求，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 这个需求是真实存在的，并且对于本项目来说是必要的
- [X] 我已经提供了该需求的清晰、详细、完整的方案，请按该方案进行实施
- [X] 我认为我提供的信息已经足够，不需要其他人继续完善该需求的其他细节
- [X] 我同意跟进该需求的实施进程，并在适时提供合理的解释与纠正

NoNamedCat, 2023-11-19:
The "hello world" aka "Blink" sketch in the Arduino examples works fine to me. What do you want? Can you be more specific on your request please?

HalfSweet, 2023-11-29:
Now we have an official wiki. you can find some potentially useful information on it, and if you have any questions, you are welcome to ask directly :)

closed 2023-11-29 by HalfSweet

HalfSweet, 2023-11-30:
Which website are you visiting? You should use the <arduino.py32.org>

NoNamedCat, 2023-12-03:
Ohh I'm sorry, I was visiting the site that is on the Readme of this repo: "https://arduino.py32.halfsweet.cn/"