py32duino/Arduino-PY32 #2: "This header is not available in freestanding mode."

NoNamedCat, 2023-11-30:
### 描述一下这个bug / Describe the bug

when I try to compile the simple "blink" program the output says:

.arduino15/packages/PY32Duino/tools/xpack-arm-none-eabi-gcc/13.2.1-1.1/arm-none-eabi/include/c++/13.2.1/bits/requires_hosted.h:34:4: error: #error "This header is not available in freestanding mode."
   34 | #  error "This header is not available in freestanding mode."
   

### 复现步骤 / To Reproduce

v0.1.1
System: Linux Mint

### 如果正常，应该是什么样 / Expected behavior

Compile the sketch.

### 截图 / Screenshots

[image]

### 日志 / Logs

.arduino15/packages/PY32Duino/tools/xpack-arm-none-eabi-gcc/13.2.1-1.1/arm-none-eabi/include/c++/13.2.1/bits/requires_hosted.h:34:4: error: #error "This header is not available in freestanding mode."
   34 | #  error "This header is not available in freestanding mode."

### 系统 / System

Mint 22.04

### PACK包版本 / Version

0.1.1

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

HalfSweet, 2023-12-01:
Fixed in 0.1.2

NoNamedCat, 2023-12-01:
Sorry. With the 0.1.2 i have this issue:

WARNING: library SrcWrapper claims to run on AirMCU architecture(s) and may be incompatible with your current board which runs on PY32 architecture(s).
In file included from /home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/system/PY32F0xx/py32f0xx_hal_conf_default.h:220,
                 from /home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/system/PY32F0xx/py32f0xx_hal_conf.h:13,
                 from /home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Inc/py32f0xx_hal.h:33,
                 from /home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/system/Arduino-PY32F0xx-Drivers/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:195,
                 from /home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/cores/arduino/py32/py32_def.h:26,
                 from /home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/cores/arduino/py32/clock.h:19,
                 from /home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/cores/arduino/wiring_time.h:23,
                 from /home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/cores/arduino/wiring.h:38,
                 from /home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/cores/arduino/Arduino.h:36,
                 from /tmp/arduino/sketches/2C45634B9C38D38B552DD500A30E66FE/sketch/sketch_dec1a.ino.cpp:1:
/home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Inc/py32f0xx_hal_dma.h:116:3: error: 'DMA_Channel_TypeDef' does not name a type; did you mean 'DMA_InitTypeDef'?
  116 |   DMA_Channel_TypeDef   *Instance;                                                    /*!< Register base address                  */
      |   ^~~~~~~~~~~~~~~~~~~
      |   DMA_InitTypeDef
/home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Inc/py32f0xx_hal_dma.h:136:3: error: 'DMA_TypeDef' does not name a type; did you mean 'TIM_TypeDef'?
  136 |   DMA_TypeDef          *DmaBaseAddress;                                               /*!< DMA Channel Base Address                */
      |   ^~~~~~~~~~~
      |   TIM_TypeDef
In file included from /home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/system/PY32F0xx/py32f0xx_hal_conf_default.h:284:
/home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Inc/py32f0xx_hal_rtc.h:173:3: error: 'RTC_TypeDef' does not name a type; did you mean 'RCC_TypeDef'?
  173 |   RTC_TypeDef                 *Instance;  /*!< Register base address    */
      |   ^~~~~~~~~~~
      |   RCC_TypeDef
/home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/cores/arduino/py32/backup.h: In function 'void resetBackupDomain()':
/home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Inc/py32f0xx_hal_rcc.h:1251:60: error: 'RCC_BDCR_BDRST' was not declared in this scope
 1251 | #define __HAL_RCC_BACKUPRESET_FORCE()   SET_BIT(RCC->BDCR, RCC_BDCR_BDRST)
      |                                                            ^~~~~~~~~~~~~~
/home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/system/Arduino-PY32F0xx-Drivers/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:170:42: note: in definition of macro 'SET_BIT'
  170 | #define SET_BIT(REG, BIT)     ((REG) |= (BIT))
      |                                          ^~~
/home/nigrum/.arduino15/packages/PY32Duino/hardware/PY32/0.1.2/cores/arduino/py32/backup.h:71:3: note: in expansion of macro '__HAL_RCC_BACKUPRESET_FORCE'
   71 |   __HAL_RCC_BACKUPRESET_FORCE();
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

HalfSweet, 2023-12-01:
What development board model did you choose?

NoNamedCat, 2023-12-01:
PY32F002

NoNamedCat, 2023-12-01:
When I choose the PY32F030 i have another problem:

WARNING: library SrcWrapper claims to run on AirMCU architecture(s) and may be incompatible with your current board which runs on PY32 architecture(s).
/home/nigrum/.arduino15/packages/PY32Duino/tools/xpack-arm-none-eabi-gcc/13.2.1-1.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino/sketches/2C45634B9C38D38B552DD500A30E66FE/sketch_dec1a.ino.elf section `._user_heap_stack' will not fit in region `RAM'
/home/nigrum/.arduino15/packages/PY32Duino/tools/xpack-arm-none-eabi-gcc/13.2.1-1.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 304 bytes
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

This is with the blink example.

HalfSweet, 2023-12-02:
When choosing the F003 or F030 models with 2k RAM, it is possible that the linker will complain about not having enough RAM space. This problem is due to the fact that the framework takes up too much of it... and the main Serial library takes up quite a bit.

We can switch to the `Disabled (no Serial support)` option in the `U(S)ART support` settings, or modify the buffer size by manual injection, which will help solve the problem.

HalfSweet, 2023-12-02:
In version 0.1.3, PY32F002A should compile properly.

Feel free to contact me if you have any problems with the runtime, and any comments you can make about this project would be greatly appreciated.

garudaonekh, 2023-12-02:
Yes, it works with PY32F002 now.

NoNamedCat, 2023-12-04:
Here is the schematic for a board that I'm designing. What do you think? Are you interested on the design?
[image]

HalfSweet, 2023-12-04:
> Here is the schematic for a board that I'm designing. What do you think? Are you interested on the design? ![PY-CAT-002](https://private-user-images.githubusercontent.com/39467683/287611411-9d7f05cf-fa3e-4a13-bd49-e6de24cf575a.svg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE2NzI1MzUsIm5iZiI6MTcwMTY3MjIzNSwicGF0aCI6Ii8zOTQ2NzY4My8yODc2MTE0MTEtOWQ3ZjA1Y2YtZmEzZS00YTEzLWJkNDktZTZkZTI0Y2Y1NzVhLnN2Zz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA0VDA2NDM1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYyNWE0ZTEwMGUzMGYxMmMxNzFmYTBhYTgwYzE2MDQzMTM4ZDI0MzBhYjU1OGM3OGVjOWM5ODJkMjhkODI2MDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.JYwIJL2IQ67TcQx9h57jLdcppMaRj8xTxXgtghkCIb8)

One thing worth noting is that the PY32 doesn't necessarily interact well with 3.3V level peripherals - either UARTs or SWDs - when it's running on a 5V supply, which I think is kind of a pitfall.

And, the V3 pin of the CH340 seems to require an external 0.1uF capacitor, I'm not quite sure, you'll need to check the datasheet