py32duino/Arduino-PY32 #9: Add  a diagram of connections to program  PY32F002

vinorasu, 2024-02-06:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

I am unable to program the sop8  PY32F002AL15S6TU. using the board "Generic PY32F002A Series"

I tested the usb to serial connected to  PA1 PA2 for serial port ,but I am unable to program the chip from arduino 2.0 with this core.
Is there a Pin to drive to ground or VCC to program the chip?

There is no option to add an STlink programmer to use   PA14/PA13   for SWD  protocol.

### 描述您想要的解决方案 / Describe the solution you'd like

Please provide a diagram of connections to program the chip, or instructions how to program it 
or add an STlink programmer ,thank you.

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

To be able to select a STlink programmer in the arduino menu or to use the serial connection when the connections diagram or instructions are provided.

### 确认信息

- [X] 检查过该需求，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 这个需求是真实存在的，并且对于本项目来说是必要的
- [X] 我已经提供了该需求的清晰、详细、完整的方案，请按该方案进行实施
- [X] 我认为我提供的信息已经足够，不需要其他人继续完善该需求的其他细节
- [X] 我同意跟进该需求的实施进程，并在适时提供合理的解释与纠正

vinorasu, 2024-02-08:
[image]

and the PDF where I got the pinout  https://datasheet.lcsc.com/lcsc/2303271200_PUYA--PY32F002AL15S6TU_C5292060.pdf
can this SOP8 or the SOP16 be programmed with this core? 
Thanks in advance

vinorasu, 2024-02-14:
No worries and thanks for your reply.
In case you decide to implement the support for  SWD programming with ST-link , this is what worked for me with pyocd:

**pyocd load ./Blink.ino.hex -t py32f002ax5 --config ./pyocd.yaml**
	0000469 I Loading /home/javier/Descargas/PUYA-SEMI PY32F002/PY32Duino.PY32.GenF002Acompiled/Blink.ino.hex [load_cmd]
	[==================================================] 100%
	0002038 I Erased 12288 bytes (3 sectors), programmed 9088 bytes (71 pages), skipped 0 bytes (0 pages) at 5.68 kB/s [loader]
 
and the content of the config file being listed with command  **cat pyocd.yaml** 
pack:
- ./Puya.PY32F0xx_DFP.1.1.0.pack

That file "Puya.PY32F0xx_DFP.1.1.0.pack"  comes from this rar file: https://www.puyasemi.com/uploadfiles/2022/11/PY-MCU%E8%B5%84%E6%96%99-20221117.rar

trefase, 2024-02-28:
Can you give more details on how you got it working @vinorasu ?