py32duino/Arduino-PY32 #16: can not compile in arduino ide

be-engineer, 2024-05-04:
### 描述一下这个bug / Describe the bug

I installed py32duino in arduino ide and selected py32f003 board，when I compile  blink demo code.I got many error messages

### 复现步骤 / To Reproduce

Open demo code ：
```
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);                      // wait for a second
}
```
Select board Generic PY32F003 series.
Press compile button.

### 如果正常，应该是什么样 / Expected behavior

If OK, the code should compile without errors.

### 截图 / Screenshots

[image]

### 日志 / Logs

error message are as follow:
```
WARNING: library SrcWrapper claims to run on AirMCU architecture(s) and may be incompatible with your current board which runs on PY32 architecture(s).
In file included from /home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Inc/py32f0xx_hal_def.h:35,
                 from /home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Inc/py32f0xx_hal_rcc.h:32,
                 from /home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/PY32F0xx/py32f0xx_hal_conf_default.h:208,
                 from /home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/PY32F0xx/py32f0xx_hal_conf.h:13,
                 from /home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Inc/py32f0xx_hal.h:33,
                 from /home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_rcc.c:57,
                 from /home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/libraries/SrcWrapper/src/HAL/py32yyxx_hal_rcc.c:8:
/home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_rcc.c: In function 'HAL_RCC_DeInit':
/home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_rcc.c:251:28: error: 'RCC_CR_PLLRDY' undeclared (first use in this function); did you mean 'RCC_IT_PLLRDY'?
  251 |   while (READ_BIT(RCC->CR, RCC_CR_PLLRDY) != 0U)
      |                            ^~~~~~~~~~~~~
/home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:174:41: note: in definition of macro 'READ_BIT'
  174 | #define READ_BIT(REG, BIT)    ((REG) & (BIT))
      |                                         ^~~
/home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_rcc.c:251:28: note: each undeclared identifier is reported only once for each function it appears in
  251 |   while (READ_BIT(RCC->CR, RCC_CR_PLLRDY) != 0U)
      |                            ^~~~~~~~~~~~~
/home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/CMSIS/Device/PY32F0xx/Include/py32f0xx.h:174:41: note: in definition of macro 'READ_BIT'
  174 | #define READ_BIT(REG, BIT)    ((REG) & (BIT))
      |                                         ^~~
/home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_rcc.c:260:8: error: 'RCC_TypeDef' has no member named 'PLLCFGR'; did you mean 'CFGR'?
  260 |   RCC->PLLCFGR = 0x00000000u;
      |        ^~~~~~~
      |        CFGR
/home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_rcc.c: In function 'HAL_RCC_OscConfig':
/home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Inc/py32f0xx_hal_rcc.h:1514:56: error: 'RCC_TypeDef' has no member named 'PLLCFGR'; did you mean 'CFGR'?
 1514 | #define __HAL_RCC_GET_PLL_OSCSOURCE() ((uint32_t)(RCC->PLLCFGR & RCC_PLLCFGR_PLLSRC))
      |                                                        ^~~~~~~
/home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_rcc.c:322:21: note: in expansion of macro '__HAL_RCC_GET_PLL_OSCSOURCE'
  322 |     temp_pllckcfg = __HAL_RCC_GET_PLL_OSCSOURCE();
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Inc/py32f0xx_hal_rcc.h:1514:66: error: 'RCC_PLLCFGR_PLLSRC' undeclared (first use in this function)
[... 131 lines cut ...]
 1275 |   RCC_OscInitStruct->PLL.PLLSource = (RCC->PLLCFGR & RCC_PLLCFGR_PLLSRC);
      |                                            ^~~~~~~
      |                                            CFGR
/home/leon/.arduino15/packages/PY32Duino/hardware/PY32/0.1.3/system/Arduino-PY32F0xx-Drivers/PY32F0xx_HAL_Driver/Src/py32f0xx_hal_rcc.c:1275:54: error: 'RCC_PLLCFGR_PLLSRC' undeclared (first use in this function)
 1275 |   RCC_OscInitStruct->PLL.PLLSource = (RCC->PLLCFGR & RCC_PLLCFGR_PLLSRC);
      |                                                      ^~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

### 系统 / System

debian12

### PACK包版本 / Version

0.1.3

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

HalfSweet, 2024-05-04:
What specific series did you choose? PY32F003x8 or PY32F003x6

be-engineer, 2024-05-10:
> What specific series did you choose? PY32F003x8 or PY32F003x6
PY32F003x8

XusMochizuki, 2024-05-12:
same here, i tried build a f002(air001) led blink rom and download to 003, download can be success, but seems nothing works :(

XusMochizuki, 2024-05-12:
Hi. I found a way to resolve this problem and its working now. 
you need change the include file in 
> Arduino15\packages\PY32Duino\hardware\PY32\0.1.3\system\Arduino-PY32F0xx-Drivers\CMSIS\Device\PY32F0xx\Include\py32f0xx.h

about line 126(maybe, i change some code thus i dont know the origin line number :D, but its close)

replace 
`#elif defined(PY32F003x8)
#include "py32f003x8.h"`

to 
`#elif defined(PY32F003x8)
#include "py32f030x8.h"`

then compile and run will be ok just like other chips.

with error logs, i think thats because py32f003x8.h doesn't  contain relevant defines about "Bit definition for RCC_CR register "
and I didnt find it in official latest py32f003x8.h, but py32Arduino use the generic codes for all chips that need some definitions like RCC_CR_HSION doesn't contain in 003xx.h

after change 003x8 to 030x8 , seems all function works correct and u can use all of 64k rams
i tested some codes like LED blink, Serial print and DHT11 Temper and Humid sensor, it runs successfully

btw this is my code about DHT11
`#include <DHT11.h>

DHT11 dht11 = DHT11(PA_12);
void setup() {
    Serial.begin(9600);
    Serial.println("ARDUINO PY003 READY!");
}

void loop() {
  int humid = dht11.readHumidity();
  int temper = dht11.readTemperature();
  Serial.printf("Temp:%d Humid:%d\n",temper,humid);
  delay(1000);
}
`

oh, and the wire connect: 
USBTTL -> py32f003f18P6
TXD -> PA3
RXD -> PA2

pull up boot(maybe button or pin header) and reset to download mode :D

hope someone can helps to fix this bug

HalfSweet, 2024-05-12:
Thank you very much for your reply, may I ask what is the real model of the MCU you are using @XusMochizuki

XusMochizuki, 2024-05-12:
> Thank you very much for your reply, may I ask what is the real model of the MCU you are using @XusMochizuki

py32f003f18P6

be-engineer, 2024-05-23:
> > Thank you very much for your reply, may I ask what is the real model of the MCU you are using @XusMochizuki
> 
> py32f003f18P6

It works fine with modified .h file,thanks

NoNamedCat, 2024-06-07:
Can you make a pr with the modified code?

ANH2018, 2024-06-09:
Dear @XusMochizuki 
Do you have any tool to upload code? 
I wrote a program using Arduino ide but I don't know how to load the program. Thanks

be-engineer, 2024-07-12:
> Dear @XusMochizuki Do you have any tool to upload code? I wrote a program using Arduino ide but I don't know how to load the program. Thanks

I use pyocd with pwlink tool in linux.

XusMochizuki, 2024-07-12:
> Can you make a pr with the modified code?

Im afraid that I cant do it now, thats because this proj was created based some MCU with PLL function, many codes need reference the defines of PLL, these defines in the official file such as "py32f003x8.h", due to issues such as open source protocols not being included in this GitHub project, it is not feasible to add definitions. Although it is possible to modify the code in this repository for areas that require special definitions (such as # ifdef NOPLL...), this is a complex project and I currently do not have much time to solve this problem. Therefore, if conditions permit, I recommend fixing it by modifying the reference files

我很抱歉我现在不能做到, 因为项目是基于有PLL外设的MCU的代码开发的, 在像是py32f003这样的mcu代码中并不包括这些定义, 并且定义代码是由普冉官方创建的, 由于开源协议等问题并不包括在该github项目中, 因此通过添加定义的方法并不可行, 虽然可以通过修改该仓库中的代码对需要特殊定义的地方进行修改(例如#ifdef NO_PLL...)但这是一个较为复杂的工程, 由于我最近的工作较为繁忙因此难以抽出较多时间来修复 因此, 如果条件允许, 我还是推荐通过修改引用文件的方式来修复