py32duino/Arduino-PY32 #22: The examples for generic PY32F030 not working for me.

61pete, 2024-12-24:
Arduino IDE 234, tried I2C scanner in the WIRE folder which works well on my Mega 2560, but does nothing on my PY32F030EK28.

The MCU is on a disposable vape(Nasty Bar 20000) which Im connecting to with a FT232 TTL cable direct to chip.

I2C example compiles and uploads fine, but nothing shows in the serial monitor.

Ive also tried the SevSeg.ino sketch, but the wrong elements/segments light up on the PY32F030

HalfSweet, 2024-12-25:
Sorry, can you provide more information such as your code or some relevant log information to assist in the determination?
If there is no output, is it possible that the MCU is still in reset or the baud rate of the UART is not correct? You can try a hard reset by pressing the RST button to see if there is some output.

61pete, 2024-12-25:
The code is the i2c example from generic PY32 board folder.

#include <Wire.h>

void setup() {

  Serial.begin(9600);
  Wire.begin();
  Serial.println("\nI2C Scanner");
}

void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++) {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.

    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    
    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address < 16) 
        Serial.print("0");
      Serial.println(address, HEX);

      nDevices++;
    }
    else if (error == 4) {
      Serial.print("Unknown error at address 0x");
      if (address < 16) 
        Serial.print("0");
      Serial.println(address, HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found");
  else
    Serial.println("done");

  delay(5000);           // wait 5 seconds for next scan
}

/////////////////////////////
The output, success with uploading.

WARNING: library Wire claims to run on AirMCU architecture(s) and may be incompatible with your current board which runs on PY32 architecture(s).
WARNING: library SrcWrapper claims to run on AirMCU architecture(s) and may be incompatible with your current board which runs on PY32 architecture(s).
Sketch uses 19568 bytes (29%) of program storage space. Maximum is 65536 bytes.
Global variables use 1332 bytes (16%) of dynamic memory, leaving 6860 bytes for local variables. Maximum is 8192 bytes.
"C:\Users\pete\AppData\Local\Arduino15\packages\PY32Duino\tools\AirISP\1.2.8.0/AirISP.exe" --chip "auto" --port "COM15" --baud 115200 --before default_reset --after hard_reset write_flash -e -p 0x08000000 "C:\Users\pete\AppData\Local\arduino\sketches\AE2A9F7B652C0DC2CD13F8F612292FB7/i2c_scanner.ino.bin"
AirISP v1.2.8.0
Serial port COM15
Connect...
Chip PID is: 0x04 0x40 
Erasing flash (this may take a while)...
Erase flash sucess, in 73,6155 ms.
start write data ...
Write 19840 bytes at 0x08000000 in 16802,8925 ms

Leaving...
Hard resetting via RTS pin...
//////////////////////////////////////

A print screen showing nothing happening via serial, correct port COM15 at 9600 baud.

[image]

1. I thought that perhaps the WIRE.h library was designed for AVR and not ARM like the PY32 is looking for.
2. The other thing is that NASTY JUICE CO, the maker of the vape pcb might have got a custom pin map from PUYASEMI.

61pete, 2024-12-27:
I finally got the I2C scanner working on another vape, a BLVK Bar 20000, 
Turns out the MCU was sending the info to UART PA2 & PA3, while I was connected to PA14 & PA15.
I have an OLED on pins 2 & 3 ie PF0 & PF1 and it isnt detecting it, so perhaps I'll have to move the SDA & SCL  to another place on the MCU.

closed 2024-12-27 by 61pete

NoNamedCat, 2024-12-28:
Sorry, can you show the pcb of that vape and the connection to the microcontroller? Can you control leds or something in the vape hardware? I'm very curious about this

61pete, 2024-12-28:
Can do no probs.
Basically its a disposable vape with a PY32F030EK28 in a QFN32 package for the MCU. 
[image]

Of the 15 dispo's Ive pulled apart, only the Nasty Bar 20000 and BLVK Bar 20000 use the PY32F030EK28 ic's.
If you check out the F030 datasheet, you will see that 4 digit tubes and 7 segments are multiplexed by the PY32 chip.

The leds side, minus the screen inlay so I can get to the pads for BOOT0 and I2C.
[image]

This is the ic side where Ive pulled off the USB daughter board so I can add a switch between it and the mcu.
That way the battery can stay connected to the charging part of the circuit without the mcu drawing power.
[image]

The only other one with the PY32 ic is the BLVK Bar 20000 puffs.
The via's are much bigger and easier to slide a enameled copper wire through to reinforce them. 
I forget how many pads Ive lifted off the Nasty pcb's :)
On the left are the PA2/PA3 UART connections and on the right end the BOOT0 and + volts rail to pull BOOT0 high to flash code.
[image]

A view of the IC side, Ive already pulled off some components off as well as the batt to make it easier to trace the layout.
[image]

Both of these vapes have a 13500 li-ion 800mah batt as standard.

I first tried lighting the digit tubes with Arduino IDE but the wrong segments were being lit.
[image]

Turns out from the datasheets that Nasty Juice Co dont use the same segments as PUYA documented.
The segments should be default on PA0 to PA7, but PA7 is the DP seg which I dont use.
Nasty split the segs with 3 on PA, 3 on PB and 1 on PF4 BOOT0, so you can see how that would create problems.
Especially since Im only on CH3 of "Beginning  C for Arduino by Jack Purdun" :)

I've since managed to get it working on a 128x64 OLED, but thats like a 2nd place win. lol

NoNamedCat, 2024-12-28:
Very interesting device. I think that you could repropose the device to measure something like a voltage meter or something like that.

61pete, 2024-12-29:
I dont think so, but I may be wrong.
What I understand is that its ADC range is from 0 to VCC (5v).
You'd have to go through the datasheets and reference manual to check though.